pixhawk/Hardware #18: Settle connector standards with Dronecode

LorenzMeier, 2015-11-05:
Active discussion on Dronecode mailing list.

LorenzMeier, 2015-11-05:
Most manufacturers have settled for JST SH (what APM 2.x used for GPS and what OpenPilot / Naze, etc seem to use) for that use case. Its cheap, available and “good enough” for a QAV 250 small FPV racing quad. Trying to establish something else seems like an uphill battle, since there is a “standard” already.

For the larger systems I have seen a mix of DF13, JST GH and Molex Clik-Mate. The JST GH is a Clik-Mate equivalent and it should fit the same footprint and plastics:
http://www.jst-mfg.com/product/pdf/eng/eGH.pdf

LorenzMeier, 2015-11-05:
You can compare the 6-pos GH vs the 6-pos Clik-Mate - the deviations I found are in the 0.1 mm range. Footprint seems to match as well, but better check yourself. I checked the vertical footprint we use most.

http://www.jst-mfg.com/product/pdf/eng/eGH.pdf

http://www.molex.com/pdm_docs/sd/5023820671_sd.pdf

LorenzMeier, 2015-11-05:
Molex Micro-Lock would be tiny, but is as expensive as Clik-Mate:
http://www.molex.com/molex/products/family?key=microlock&channel=products&chanName=family&pageTitle=Introduction

pavel-kirienko, 2015-11-05:
For reference, side-entry 4 pin 1.25mm:

* [Molex Micro-Lock](http://www.digikey.com/product-detail/en/5041950470/WM10585CT-ND/4879346)
* [Molex CLIK-Mate](http://www.digikey.com/product-detail/en/5023820470/WM4781CT-ND/2649175)
* [JST GH](http://www.digikey.com/product-detail/en/SM04B-GHS-TB(LF)(SN)%3B-/455-1566-1-ND/807834)

DigiKey is rather irrelevant if large quantities are of interest, but still serves as a good reference. It can be seen that JST GH is far cheaper indeed, and all connectors are stockable even in small quantities.

Using this data and the fact that a certain large manufacturer is reportedly using JST GH, I would pick it, but it's hard to make a decision without studying the options up close.

proficnc, 2015-11-05:
As things have "changed" in regards to the manufacturing here... I am no longer against a change in connector type.  However, the reasons we wanted to move to clickmate included the following...

1. Easier to remove.
2. Secure.

If these goals can be met, then I am happy...

pavel-kirienko, 2015-11-05:
@proficnc I wouldn't even consider an option without some sort of positive lock, luckily JST GH has it so I consider it secure and generally usable (but again, I didn't have a chance to tinker with it myself so I can only guess).

proficnc, 2015-11-05:
Yeah, it looks good...

j3qq4hch, 2015-11-05:
Some pics for comparing
[image]
[image]

j3qq4hch, 2015-11-05:
Source sketchup model can be found here:
ftp://pkir.net/CONNECTOR_COMPARE.skp

j3qq4hch, 2015-11-05:
Looks like JST wins

pavel-kirienko, 2015-11-05:
Thanks Alexander. So far JST GH definitely wins, we're going to order some for evaluation and then I'll report back.

pkocmoud, 2015-11-05:
Nick has tried the GH connector and here are his comments:

-this is the same concept as Molex ClikMate.....the same metal pin, but the better pin guides. To me the GH connector design is better than ClikMate

-the plastic body of the GHs is better designed

-I discussed these GH connectors with Georgi Kasabadjakov - the guru in mold production and his opinion is like mine.

nickarsov, 2015-11-05:
Hi guys,
The GH has one main benefit ( in addition to the better design ) - the metal pins have guides....take a closer look at them and you'll see the better design. Bye bye Molex.....

LorenzMeier, 2015-11-10:
Final vote, please participate: http://doodle.com/poll/ynkumy4u8vrfhhme

pavel-kirienko, 2015-11-10:
Lorenz, I think calling it final might be a bit premature. I don't think all of the people involved in the voting are making informed enough decisions, rather I would expect many participants to vote for their own precious connector type in order to avoid the hassle of transitioning to the new type.

This poll indeed seems like a good indicator of the community/industry's preferences, but I wouldn't trust it to make the final decision.

LorenzMeier, 2015-11-10:
The disclaimer in the TSC announcement said exactly that - it is an additional data point. The final is meant to indicate that this is my last attempt to solicit feedback before making a decision.

ivereninov, 2015-11-11:
Connectors always have been a very tough decision for me. As engineers we always aim for the smallest connector, to make our devices smaller (==cooler),  but they are not always optimal for end users. We feel comfortable with them, because we use them everyday and we adapt. 

Some comments about presented options:
1. Still not easy to handle, try to reconnect some of them in -10C during a mission when your fingers freeze.
2. Only 1A in power (in CAN topology that is for all device in the net), limits powering through CAN.
3. Impossible to have nice cable overmolding and cable stress reliefs.
4. None of these 1.25mm connectors are meant to be used for module to module communication, mostly they are designed for use inside consumer electronics enclosures. 
5. They have very low resource, mostly 10-20 connections. Design goal is that they survive assembly on the factory and maybe disassembly during service. Not intended to be disconnected often. You can have removable wings or payloads.

There is definitely a need in a tiny connector and JST GH looks like clearly winner here, but most installations nowadays would not require miniaturization as APM is mostly chosen for flexibility, thus more payloads and larger frames.

I think there was enough user feedback about DF13 to consider using a larger connector(2.54mm for example)?  

As a compromise, how about we define several types of connectors (Dronecode connector, Dronecode micro connector)? 
 
Since most autopilots will be getting rid of numerous SPI, I2C, ADC connectors in favour of CAN there will be plenty of space left for a larger connector.

LorenzMeier, 2015-11-24:
@ivereninov We will cross that bridge when we get there. I expect that Dronecode will eventually have three tiers:

  * Consumer connectors. Something like JST SH
  * Industrial in-module connectors. JST GH
  * Industrial vehicle-level connectors. Automotive and sealed

We have set the middle tier now and we'll leave the bottom and top tiers open until we *really* have adopted CAN.

proficnc, 2015-11-24:
To be honest, a recommendation here is ok, but the application diversity of these platforms calls for flexibility 

Can is Clickmate.... All current can hardware developers and manufacturers agree... And as its new, that's cool to be a "standard" but everything else has other things to concider.

pavel-kirienko, 2015-11-25:
@proficnc I think it's not too late to move CAN to the new connector, bridging the transition with adapter cables. Board manufacturers can also win from the fact that this new connector is footprint compatible with CLIK-Mate.

pavel-kirienko, 2015-11-25:
I have evaluated JST GH and couldn't find any issues with them. I agree with Nick's assessment above.

proficnc, 2015-11-25:
IMHO, I actually think there are significant benefits to having the CAN connector be different to the other connectors.  It makes it very clear to the consumer what each connector is.

For Pixhawk 2, we are seriously concidering moving back to DF 13's for supply chain reasons, but keeping clickmate for CAN.

I personally love the idea that there would be a standard for certain connectors, and do not object that it is GH. But I think that the realities of usage have not yet really been discussed here.  

Why would we make the CAN and I2C connectors the same? By using different connectors, we make it simple for users.

I actually think we need a real meeting on this topic, and to discuss it calmly

LorenzMeier, 2016-01-14:
Thanks everybody for contributing! The first boards with JST GH are fielded now and the connector is the best I've used in years. I'm going to stop lobbying for it now, because any of you working with it will be immediately convinced its totally worth it.

closed 2016-01-14 by LorenzMeier

HackInvent, 2016-07-19:
Hi, am sure my question comes late, why didn't we consider Molex Picoblade series in the discussion?
http://www.molex.com/pdm_docs/sd/530470410_sd.pdf

d-skinner, 2016-07-22:
Hi All,

I'm interested in how the dronecode wire colours were selected? Why are we using black for everything that's not VCC??? I know this as be a "standard" for a while but it's always been a little scary to me. It seems a lot nicer to use red for VCC, black for GND and maybe white for everything else (ie. signals)? I only say this from my experience over the years of development that when you need to quickly hook up some electronics for testing it can be useful to simply cut a cable and solder the wires on. Even though you're still going to need to check what's what with the white wires, it would at least make it clear that you're hooking up a power wire (either VCC or GND) rather than a signal wire.

Thoughts?