pixhawk/Hardware #21: Pixracer: Minimal, ultra-low-cost Pixhawk variant for small drones

LorenzMeier, 2015-11-05:
Working on this with Phillip K. and Nick. Purposefully reduced to the bare minimum to allow a clear product-market-fit for racer pilots. Has CAN in contrast to small other variants like Pixfalcon. Will fill this with feedback and issues as testing becomes available.

kwikius, 2016-05-13:
Is the CAN cable intended to chain in and out of all the ESC's? My personal view is that makes this type of CAN bus architecture too cumbersome for small drones. A star architecture is a more natural fit

LorenzMeier, 2016-05-13:
Its up to you to use a star cabling or not. The stub length certainly allows it.

closed 2016-05-13 by LorenzMeier

kwikius, 2016-05-13:
OK.. What is the maximum allowed stub length ( or rough guide to that)

LorenzMeier, 2016-05-13:
CAN standards says 0.6 m per stub, so unless you build a monster drone easy to comply with.

kwikius, 2016-05-13:
OK it is interesting information that it is allowed. My use case was to put arbitrary things, sensors, RC receiver, GPS etc in the wings of an aircraft. The idea is that for high performance soaring that would allow a slimmer fuselage. Anyway I guess this will work fine for racing quadcopters.