pixhawk/Hardware #24: Foamless IMU

proficnc, 2015-11-05:
With the move to the MPU9250, and the potential for high rate reading. The need for a large isolation system is reduced.
Some form of mechanical and thermal isolation is still needed, but we should consider some options here.

pkocmoud, 2015-11-09:
Hi Philip. The biggest complaint I hear about the MPU9250 is the internal magnetometer is difficult to use or is not reliable. The XRacer has a MPU9250 (Gyro/Accel/Mag) + ICM20608 (Gyro/Accel) / HMC5983 (Mag). I am hoping that this will give us the opportunity to validate the usability of the MPU9250. It should be noted that the ICM20608 is a newer version on the MPU6500.

proficnc, 2015-11-10:
Yes, we have the an HMC5983 next to the 9250... So it will be interesting... 

Yes, and if the mag in the 9250 doesn't do what we need, the ICM is the logical replacement

pkocmoud, 2016-11-08:
Chatted with @proficnc and this issue is obsolete.

closed 2016-11-08 by pkocmoud