pixhawk/Hardware #29: Settle on standard CAN transceiver

LorenzMeier, 2015-11-24:
We should identify the default part.

pkocmoud, 2015-11-24:
When developing the CANESC 1.6 it was suggested that we choose a part which was 
ISO 11898-2 compliant. https://en.wikipedia.org/wiki/CAN_bus#Standards Therefore we settled on the Linear LTC2875 http://www.linear.com/product/LTC2875 . It seems to work fine.

nickarsov, 2015-11-24:
Also a very good solution with best performance/price is TJA1051TK3/118

LorenzMeier, 2015-11-24:
@CarlOlsson Sounds like the TJA is your part then.

proficnc, 2015-11-24:
Yes, this is the Transceiver that I am using....

CarlOlsson, 2015-11-25:
Ok perfect! Thanks for the info

CarlOlsson, 2015-11-25:
I cant find the TJA1051TK3/118 in the [NXP eagle library](http://www.element14.com/community/docs/DOC-64270/l/nxp-cad-library-for-cadsoft-eagle-software) or the [pixhawk2.lbr](https://github.com/PX4/Hardware/blob/master/pixhawk2.lbr) or in https://github.com/pixhawk/Hardware/tree/master/eagle/libraries/eagle.

If I have not missed it I will create the part but where shall I add it to make it easier for future designers?

LorenzMeier, 2015-11-25:
Pixhawk 2 libraries. The current generation boards are done in Altium.

CarlOlsson, 2015-11-25:
Ok thx!

pavel-kirienko, 2015-11-25:
TJA1051 is good, but in certain cases it is preferable to use a 3.3V transceiver. In this case I would recommend [SN65HVD232](http://www.ti.com/product/sn65hvd232) - it is cheap and automotive grade; the downside is that it's only available in SO8 which is huge.

[LTC2875](http://www.linear.com/product/LTC2875) Phillip mentioned is another 3.3V alternative in a small package, but its cost may be prohibitive.

proficnc, 2015-11-25:
Agreed on the 3.3 vs 5v issue.

May I suggest that rather than trying to define a standard here... We just list devices that have been verified, with their strengths and weaknesses listed? 

As long as they perform the job they are required to do, the choice should be up to the needs of the designer.

We should list some tests, ie, is it non blocking when a micro sets pins to locked values etc....  So maybe add a blacklist of parts that we will not accept if proven not to be up to standard

CarlOlsson, 2015-11-26:
I agree.
The reason to why I asked in the first place was that I wanted to use a design already proven to work.
The list so far:
- Olimexino: **MCP2551-I/SN**
- Pixhawk ESC 1.6: **LTC2875**
- Pixhawk ESC 1.4b: **MAX3051EKA**
- s2740vc ESC and XRacer: **TJA1051TK3/118**

LorenzMeier, 2016-01-15:
@CarlOlsson What was the final choice?

proficnc, 2016-01-15:
The final choice was a whitelist and blacklist... As different components may suit different applications.

Anything that is deemed unsuitable should be blacklisted

nickarsov, 2016-01-15:
To me, the white list currently looks like:
TJA1051TK3/118
LTC2875

proficnc, 2016-01-15:
Agreed

LorenzMeier, 2016-01-15:
Thanks for the feedback Phil! I'll get that into the Dronecode wiki. What voltage are these designed for - 3.3V?

proficnc, 2016-01-15:
The Can side is always 5v, but the secondary on the TJA105 is 3.3v 

Need to check the others

This is the one I am using now for all new designs

LorenzMeier, 2016-05-10:
Noted here: https://wiki.dronecode.org/workgroup/connectors/start?&#can_tranceivers

closed 2016-05-10 by LorenzMeier