pixhawk/Hardware #38: Pixracer: PPM pin output resistor value

LorenzMeier, 2016-01-14:
@nickarsov @pkocmoud @kd0aij I'm tracking this here for now. We can't seem to be using 470R for the backwards direction, we need to use something with much higher resistance, e.g. 1.5K or 2.2K. We need to try it against real hardware.

Can you please attach (just drag images onto the text pane to attach) scope shots of the port?

pkocmoud, 2016-01-15:
DSM 470R
[image]

pkocmoud, 2016-01-15:
DSM Open Resistor
[image]

pkocmoud, 2016-01-15:
DSM R200 = 1.8K
(RC and Bind works)
[image]

pkocmoud, 2016-01-15:
DSM R200 = 2.2K
(RC works and Bind does not)
[image]

pkocmoud, 2016-01-15:
My suggestion is continue testing with the 1.8K resistor.

nickarsov, 2016-01-15:
So, PPM ( PWM->PPM converter ) works with both 1k8 and 2k2, but 2Y must be LOW ( FMU_RC_OUTPUT must be LOW when using with PPM/Spektrum sat ).

[image]

LorenzMeier, 2016-01-15:
As Nick says we have to re-test to make sure we can reach 3.3V and 0V. In your scope shots the signal is still at 1V, which seems a stretch for robustness.

nickarsov, 2016-01-15:
Lorenz, can you please fix the FMU_RC_OUTPUT = LOW when using PPM/Spektrum. With this fix I'm sure we can go with 1k8 without problems.

LorenzMeier, 2016-01-15:
I'll leave that to Mark to not confuse your workflow. @kd0aij Can you test that with Phil / Nick? Please re-share the analyser shots here so we can do a final review.

kd0aij, 2016-01-15:
@LorenzMeier We're seeing good logic levels on RC_INPUT with a variety of receivers and R200 as a 1.8K pulldown. But DSM binding is working only with clones, not authentic Spektrum satellites. The timing precision of the bind pulses is poor with fmuv4, but good with fmuv2. I'm confused by that, since it looks like the same method is being called in both cases (dsm.c::dsm_bind)

kd0aij, 2016-01-15:
latest test config is here: https://github.com/PX4/Firmware/issues/3513

LorenzMeier, 2016-01-16:
@kd0aij @pkocmoud @nickarsov Per my last discussion with Phil we propose to abandon S.BUS output and pairing for the first rev. Both are fringe use cases which have *not* been promised to the community or customers.

LorenzMeier, 2016-01-26:
Fixed in final HW rev.

LorenzMeier, 2016-08-04:
@pkocmoud What did you end up using in the final HW rev on populated boards? Did this get reflected into a design revision on the schematics for R12 (or whatever the one is which is in production)?

pkocmoud, 2016-08-04:
R12 this resistor is not populated. The downside is R12 boards cannot command the Spektrum satellites into bind mode. R14, the output was re-routed to not go thru the signal inverter. DNP is listed on both R12 and R14 schematics for the resistor in question.

closed 2016-08-04 by pkocmoud