pixhawk/Hardware #40: Pixracer: Package content

LorenzMeier, 2016-01-16:
@pkocmoud @nickarsov Its not clear what your customers get with each Racer. What exactly is in the box? Which list of cables? Mounting material (the racer frames do not come with any)? Foam?

pkocmoud, 2016-01-17:
1 WiFi Adapter - ESP-01 flashed with MavESP8266

11 Included Cables (All very flexible Silicone):
2 Power Cables - 150mm and 25mm
GPS cable  - 3D Robotics uBlox 5 GPS - 150mm
GPS cable  - Ready to Fly Quad uBlox m8 GPS - 150mm
Buzzer and Safety Switch - 150mm
Telemetry1 Cable (Separate 2.54mm females) - 150mm
Telemetry2 Cable (Separate 2.54mm females) - 150mm
OSD Cable (Separate 2.54mm females)  - 150mm
CAN Cable - GH to GH - 150mm
PPM / SBus Input Cable - 150mm
Spektrum Satellite Input Cable - 150mm

The Following are subject to change.

No Mounting Material is included.

No Foam is Included.

LorenzMeier, 2016-01-17:
It would be great to update the product listing with this and to be more specific from a customer perspective. E.g.: "PPM / SBus Input Cable - 150mm" should become "PPM / SBus Input Cable - 150mm, GH to standard servo connector and single pin (for RSSI)"

The GPS cable should not reference a hardware revision, instead something like: "GPS cable - 3D Robotics GPS module (separate serial and I2C plugs, DF13) - 150mm"

LorenzMeier, 2016-01-17:
What happens if I don't buy a power module from you - how do I get power? Did you consider adding a JST GH to DF13 power cable (replace the long version with that)?

LorenzMeier, 2016-01-17:
What is the "OSD" cable? What is its pinout? Which OSDs are supported?

LorenzMeier, 2016-01-26:
@pkocmoud Product listing still doesn't actually show or list in detail what is included: http://www.auav.co/product-p/xr-v1.htm