pixhawk/Hardware #41: Pixracer: Mounting material

LorenzMeier, 2016-01-16:
@kd0aij @pkocmoud @nickarsov The QAV250 doesn't include mounting material for the autopilot. So if you order it and a Pixracer you can't mount it. What were you planning in terms of mounting hardware?

pkocmoud, 2016-01-17:
The users that are buying and mounting this board will typically have their own particular type of mounting or vibration management solution. This is one way they tune their experience. If they want a recommendation from us on this topic, rest assured it will appear in an email or a forum.

LorenzMeier, 2016-01-17:
The difference with your offering with other offerings is that they indeed do not rely on the user to figure things out, but provide a default solution:
http://www.hobbyking.com/hobbyking/store/__67377__Mini_Flight_Controller_Mounting_Base_30_5mm_Naze32_KK_Mini_CC3D_Mini_APM_30_5mm_36mm_.html

LorenzMeier, 2016-01-17:
I'm also confused by your statement regarding the ACSP: You're saying its designed to be stacked, are you really assuming that your users have spacers in different heights available to them?

From a general commercial point of view buying a Pixracer and a QAV250 should result in a flying quad if you want to position yourself as a high-end western provider.

Four spacers cost about $0.40 total. How much is one of your bundled cables?

SimonWilks, 2016-01-17:
Here's a trivial start. Spacers would still be needed, alternatively a smarter case design to take the powerboard too. If a mount isn't made available at least as an option at least spacers would be good as it's not trivial to get a hold of these.
 
[image]

Mounted ready to go into a mini flying wing:
[image]

Sketchup and STL: [pixracer_mount.zip](https://github.com/PX4/Hardware/files/93387/pixracer_mount.zip)