pixhawk/Hardware #46: Dronecode power connector recommendations

LorenzMeier, 2016-01-17:
Since manufacturers are also doing power modules and ESCs it would be good if we could identify a low-cost automotive type we can recommend. I think the drone industrial space will need the same kind of standardisation as automotive needed.

For now just looking for recommendations, its too early to try to standardise as we hardly understand the needs and market.

@proficnc Feedback welcome!

Some common types:

  * Hobby connectors like XT60, non-crimpable
  * Professional connectors like Anderson Power Pole: http://www.andersonpower.com/us/en/products/powerpole/index.aspx (crimp able, stackable, not sealed)
  * Automotive sealed power connectors - suggestions?

pavel-kirienko, 2016-01-17:
Hobby connectors need to go (XT60, Deans T, bananas). I would propose the following set of requirements to the *standard power connector*:

- Both board-to-wire and wire-to-wire options should be available
- The connector should be crimpable for manufacturability reasons
- Positive lock
- Availability (should be available from DigiKey)
- Reasonable cost
- It is preferable to have different options for different power levels
- Small pitch

JST J300 could be a good fit for under 15A applications (which is a majority of applications); shame it's hard to stock: http://www.jst-mfg.com/product/pdf/eng/eJFA-J300.pdf

LorenzMeier, 2016-01-17:
Your requirements sound good.

Some starting points for further research (ideally the series would offer sealed and unsealed options, like some automotive series do). Or we pick a sealed and an unsealed, very cheap option (similar to JST GH).

http://www.te.com/usa-en/products/connectors/automotive-connectors/intersection/hybrid-and-electric-connecitvity-solutions.html

http://www.delphi.com/manufacturers/auto/connection-systems/standard-automotive-connection-systems

http://www.molex.com/molex/products/group?key=sealed_connectors&channel=products

LorenzMeier, 2016-01-17:
Equipment supplier pages also help, since they show what's currently being used in the industry:
http://www.waytekwire.com/products/11/Multi-Pin-Connectors/

SimonWilks, 2016-01-17:
Some thought should be given to what the "majority" would be. An F450 flying around normally for example already draws over 15A. Knowing what burst currents they can handle would be useful too.

Looking at some of Mark's logs he seems to have a steady draw of around 10-12A on a 250 already.

F450: http://logs.uaventure.com/view/GEgZgnMSKbHQMpRePVSo6m#Power_PLOT

boosfelm, 2016-01-17:
I agree to @pavel-kirienko requirements except for the current. We have been working with [Anderson Power Poles] (http://www.andersonpower.com/us/en/products/powerpole/index.aspx) for two years now and made only good experience, also meeting all the requirements stated. Depending on the crimp they are rated to 30 or 45A so it would solve for @SimonWilks as well. Also stackable and extendable is nice, 2 connectors DC and 3 from ESC to motor.
Only about price I could imagine there are cheaper solutions.

LorenzMeier, 2016-01-17:
@boosfelm Thanks! I think we can recommend the Anderson series for development vehicles, but we need to find a standard which is really cheap at quantity.

pavel-kirienko, 2016-01-17:
Let us define possible use cases before continuing. As I see it:

- **Main power supply connector**, e.g. battery to a power distribution unit. This connection obviously requires the highest current capacity. It should be either a two power pins connector, or, if we want smart batteries, two power pins plus at least three more pins for the I2C interface. The limiting factor here is probably poor availability of such combined connectors (high-power + signal pins). Wire-to-wire types are not required for this use case.
- **ESC power input connector**. Two pins, lower current capacity (this is where 15A covers vast majority of applications).
- **ESC to motor connector**. Three pins, same current capacity a above.

nickarsov, 2016-01-17:
Hi guys,
For power/signal connector combos, have a look at Molex , Amphenol and Harwin. Such connectors can't be cheap.
Some links:
http://www.molex.com/molex/products/family?key=extreme_ten60power_highcurrent_connector&channel=products&chanName=family&pageTitle=Introduction&parentKey=power_connectors
https://www.harwin.com/connector-products/browse-products/?category=Mix-Tek%20%28Power/Signal%20Off-The-Shelf%29&level=2&code=01G&count=1

LorenzMeier, 2016-01-17:
@nickarsov It doesn't have to be cheap, but cost-effective. That certainly exists in the automotive and industrial sector, because both of these are cost-sensitive.

proficnc, 2016-01-17:
The two suggestions from nick are among my favourite, other than the cost.  However, is there any reason we couldn't get a customised connector? Volume is key...

For the ESC, we are dealing with CANx2 + power, on 1 end, and 3 phase + a potential hall sensor out the other end. 

The other thing to look at, is waterproofing.  

A standard here would be nice,

pavel-kirienko, 2016-01-17:
> For the ESC, we are dealing with CANx2 + power, on 1 end

I am strongly against joining CAN with power on the same connector. First, it adds yet another connector type for CAN (we already have at least three, let's hope vendors converge to JST GH sooner than later); second, it requires manufacturing of custom harnesses because daisy chaining will not be possible.

CAN should be kept separate from power.

proficnc, 2016-01-17:
While I agree with the sentiment of what you are saying, there is a need for a waterproof, compact single connection method.

I highlight that, as there is never going to be a one size fits all here.

We need to realise that these systems are not only being used in 250 racers and DIY gear, but also in commercial Aviation grade applications.  In these applications, custom wiring looms are perfectly acceptable.

This would be highlighted in the automotive world, there is no defined CAN connector, in fact, nearly every sensor has a different connector.  While I understand that is not where we want to go, finding a connector that is designed for harsh environments, for power, and for can, is critical.  But having 4 connectors for a redundant CAN setup, is most likely not feasible for most applications.

For a general standard, the GH is fine, but for commercial exposed applications, both power and CAN must be ip68

proficnc, 2016-01-17:
For general battery power, the XT90-S is nice.

In reality, I again think that we should divide this into purposes...

Clearly Solo has a power connector that suits it's needs... We won't be attempting to standardise OEM's

What we are trying to standardise is the DIY market only, is that a safe assumption? 

From that point of view, I am all for the xt60 and xt90 power, and GH for CAN...

But let's also list options for the applications that fit outside those goals.

pkocmoud, 2016-01-18:
While the XT60 are very prevalent, I find that some become very difficult to separate and this can lead to wire / solder joint fatigue on what ever connection is serving as strain relief. I really like the Traxxas RC car connectors https://traxxas.com/products/parts/accessories/highcurrentconnector but they are proprietary and expensive.

j3qq4hch, 2017-04-20:
How about some *fir connectors from Molex? Like [megafit ](http://www.molex.com/pdm_docs/ps/PS-76823-100.pdf)or [ultrafit](http://www.molex.com/molex/products/family?key=ultrafit_power_connectors&channel=products&chanName=family&pageTitle=Introduction&parentKey=wire_to_board_connectors)

They have a lock, they have a key, they are through-hole, so very durable and robust. If 16 or 23 amps per pin is not enough - lets use several pins. They are also crimpable. The only thing I am not sure about - wire-to-wire connection with these.

EShamaev, 2017-04-26:
my few cents:

for wire-to-wire I would consider D369 series from Deutsch (TE): http://www.ttieurope.com/docs/IO/29687/TE_Deutsch_369_Series_Product_Guide.pdf

As for PCB, my choice is Harwin (while not cheap though):
https://www.harwin.com/products/M80-5000000M5-02-333-00-000/
It also comes as combined power/signal:
https://www.harwin.com/products/M80-5L10405M5-02-333-00-000/