pixhawk/Hardware #47: Pixracer R13

LorenzMeier, 2016-01-17:
@pkocmoud @nickarsov @kd0aij This is to discuss schematic revision 13.

[PIXRACER-R13.pdf](https://github.com/PX4/Hardware/files/93463/PIXRACER-R13.pdf)

LorenzMeier, 2016-01-17:
@nickarsov The issue with the S.BUS inverter (as far as I could see) is that its not bi-directional. So we wouldn't be able to pair or do S.BUS2 telemetry.

nickarsov, 2016-01-17:
The changes are on pages 2/5 and 3/5. PB0 hs been separated from PC7. 
The new labels are: PB0 = FMU_RC_INPUT_TIM and PC7 = FMU_RC_INPUT_SER
1. PPM:    RC_INPUT -> FMU_RC_INPUT_TIM
2. BIND:    PB0 becomes GPIO, HIGH for 100ms, then pulsing 3 active low pulses 50us each, then at least 10ms high, then PB0 must be either configured as input or as 3-stated
3. Spektrum/SBus are read via U1203 to FMU_RC_INPUT_SER ( PC7 ) and the inversion of the RC_INPUT signal depends on SBUS_INV ( PC13 ). Inversion is with SBUS_INV = 1.

kd0aij, 2016-01-17:
@LorenzMeier This is intended to be the optimal electrical config with abandonment of bidirectional SBUS support. I think it should allow proper DSM binding though, using PB0 as I/O

nickarsov, 2016-01-17:
Lorenz, can't we use FrSky inverter ( U1202 ) for?

LorenzMeier, 2016-01-17:
@kd0aij It just seems that we will continue to see more single-wire implementations and should position ourselves to be able to do inverted and non-inverted single wire reliably, without strings attached. For RC input we then could connect the TX pin only and use the STM32 in single-wire mode, for FrSky the cable could just pick the TX line and we can do the same in SW.

kd0aij, 2016-01-17:
I believe R13 is an interim (1 batch) solution to go into production ASAP. Right Phil?
Longer term, the question is whether both S.port and SBUS2 are both compatible with the Maxim 1-wire spec. Are there any 1-wire bidirectional buses (yet) which are non-inverted? Since we're short on GPIOs, it would be nice if the answer is no.

kd0aij, 2016-01-17:
I need to look at the STM32 USART single-wire mode. Assuming it turns the IO pin around only when transmitting it should be the ideal solution. Will hopefully be able to do that while debugging the SmartPort code.

LorenzMeier, 2016-01-17:
No you won't, because its only using the TX pin. Which will be blocked by the current unidirectional inverter. I guess you understand now what I mean.

kd0aij, 2016-01-17:
Oh, I knew that :) But I should be able to breadboard it using a telem port.

nickarsov, 2016-01-17:
@Mark: "Are there any 1-wire bidirectional buses (yet) which are non-inverted?"
Yes, there are such - DS2480B, which is an old Dallas Semi chip, but it seems it is still in production.

kd0aij, 2016-01-17:
Is that something an autopilot would need to support?

nickarsov, 2016-01-18:
Hi guys,
Here is the sch which seems to fulfill all requirements for PPM, Spektrum, Spektrum Sat, SBus I/O, S.Port, SBus2. A truth table integrated. Please check and confirm.

https://drive.google.com/file/d/0BwGTTqZqRwPLVGVqRUl2bmNHeDA/view?usp=sharing

Unfortunately I'm not allowed to upload PDF files here. Also I can't find the Pixracer folder in https://github.com/PX4/Hardware/

nickarsov, 2016-01-18:
The truth table was fixed and the link has been updated.

LorenzMeier, 2016-01-18:
As of now there is no single wire bus which is not inverted. I can see where you are going with this question. But we don't need a tri-state inverter if it has auto direction sensing.

nickarsov, 2016-01-18:
We need a non inverted Spektrum signal.

kd0aij, 2016-01-19:
Maxim's 1-wire electrical spec:
https://www.maximintegrated.com/en/app-notes/index.mvp/id/214

LorenzMeier, 2016-01-26:
Closing, as we have all this now in R14.

closed 2016-01-26 by LorenzMeier

limhyon, 2016-08-23:
I am now testing with R14, still I had a binding issue of DSMX receiver.
Any comments? Also, I couldn't find RC_DSM_BIND parameter when I use Pixracer.

pkocmoud, 2016-08-23:
What flight code and version are you testing with?

titeuf007, 2017-05-20:
hey nick
where we can download  the latest r14 file?
i would like to make my own pcb and soldering by myself but i can t arrive to find them
sorry for my english i am french