pixhawk/Hardware #49: Pixracer: Potential power sequencing issue on GPS port

LorenzMeier, 2016-01-19:
@nickarsov @pkocmoud The airspeed sensor browns out on boot. The most likely reason is that the power sequencing goes bad during boot.

It only happens when powered through battery, it doesn't happen when powered through USB.

How does the 5V rail plot look like on boot? Please do not offer any judgements, I just want to see the real data. 3 plots:

1) USB power applied
2) Battery power applied
3) A sequence of wiggling the power connector of the battery

LorenzMeier, 2016-01-19:
So it seems like connecting battery "kills" the sensor. Which seems absurd.

Maybe the device went into a bad state when we powered it reverse.

tumbili, 2016-01-19:
Correct, having usb connected the driver works. Connecting power without removing the usb cable kills the driver.

pkocmoud, 2016-01-19:
[image]

Power Up using Battery, connecting and disconnecting USB, Battery remains connected until the end.

LorenzMeier, 2016-01-19:
@pkocmoud It would be great if you could try to repro with one of these: http://www.hobbyking.com/hobbyking/store/%5F%5F62752%5F%5FHKPilot%5F32%5FDigital%5FAir%5FSpeed%5FSensor%5FAnd%5FPitot%5FTube%5FSet.html

pkocmoud, 2016-01-19:
"repro"? I have the 3DR Version to test with.

tumbili, 2016-01-19:
@pkocmoud That would be great.

pkocmoud, 2016-01-19:
[image]
Recorded while loaded with Master.

pkocmoud, 2016-01-19:
No Brown Outs Detected. The PixRacer power supply is solid during our tests.

LorenzMeier, 2016-01-19:
@tumbili If you comment out this line, does it work for you?
https://github.com/PX4/Firmware/blob/0e0c1302e41abc75e83bd3b0a324bead952f8a16/src/drivers/meas_airspeed/meas_airspeed.cpp#L228

tumbili, 2016-01-19:
@LorenzMeier same thing

LorenzMeier, 2016-01-19:
Odd. It has to be somewhere there.

LorenzMeier, 2016-01-20:
So my 2nd board has an all working sensor. We have now 3 boards:

1): 0-1.8V on the 5V rail
2): Romans board: 5V on GPS rail, but no airspeed sensor signal when powered via the 3DR power module
3): Completely working when powered at 5.3V through a lab power supply

The only thing I haven't tried yet is what happens when the sensing / ADC pins are active, as I have no power module.

I certainly have a bad board, not sure yet about the causality of the other two items.

tumbili, 2016-01-20:
@LorenzMeier Is 5.3 a magic value? Does is also work with 5V?

LorenzMeier, 2016-01-20:
Your power module is outputting 5.3V. Which is why I chose it.

nickarsov, 2016-01-20:
Yes, it works with 5V too, because the USB can't supply 5.3V

LorenzMeier, 2016-01-20:
Nick, rather than speculating it would be good if you could work out how I can return the electrically bad board to your for inspection.

nickarsov, 2016-01-20:
Lorenz, 
Speculating with what...I just answered to Roman.
And yes, you can return it back to me for inspection at:
Nick Arsov
8 Konstantin Irechek str., bl.1
4700 Smolyan
Bulgaria

LorenzMeier, 2016-01-20:
Thanks, will do! I'm closing this as it seems board-specific.

LorenzMeier, 2016-01-20:
@nickarsov the part you identified has 5V on one side and 0V on the other.

LorenzMeier, 2016-01-20:
Soldering looks sketchy. I will rework it later today or tomorrow.

LorenzMeier, 2016-01-26:
This was in fact the airspeed sensor. Still have one board with 1.8V on that port, but that is probably manufacturing QA. Replaced with: https://github.com/PX4/Hardware/issues/55

closed 2016-01-26 by LorenzMeier