pixhawk/Hardware #50: Pixracer GPS power port: 1.8V output

LorenzMeier, 2016-01-20:
@pkocmoud @nickarsov @kd0aij I'm measuring 1.8V output on the GPS power port with just the airspeed sensor attached. Can you confirm that? I'm measuring 0V with nothing attached.

nickarsov, 2016-01-20:
Lorenz, which cable do you use as GPS cable has been totally swapped - both GPS connector and I2C connector. I already notified all users about. Can you send a screenshot of the wiring?

LorenzMeier, 2016-01-20:
Its not the wiring, I measured the connector pins. I confirmed that my 2nd board provides 5V, so I got a bad board.

LorenzMeier, 2016-01-20:
Where was the diode that was the wrong way around? Can you mark it in the layout?

nickarsov, 2016-01-20:
There is just a resettable fuse between 5V and VDD_5V_PERIPH. Does it get
hot?

nickarsov, 2016-01-20:
Picture attached

nickarsov, 2016-01-20:
[image]

LorenzMeier, 2016-01-20:
Components on the working and non-working board are all oriented exactly the same.

nickarsov, 2016-01-20:
Maybe the fuse FR101 gets overloaded by some reason...
What is the voltage on its pads?

pkocmoud, 2016-01-27:
This appears to be a defective board. Please close this issue unless other examples appear.

closed 2016-11-08 by pkocmoud