pixhawk/Hardware #54: Pixracer: FrSky Telemetry cable

LorenzMeier, 2016-01-24:
@kd0aij @nickarsov Do we need to connect the 5V signal to the receiver? That might be problematic and we might want to only connect S.PORT and GND.

kd0aij, 2016-01-24:
I was not connecting 5V from pixracer to the X4R receiver in my testing, except of course via the RC input cable.
Its purpose is to supply power from the RX to sensors on the bus, so the sensors do not source 5V. The S.port + pin on my X4R-SB is connected directly to the + line on the servo bus, so there shouldn't be a problem if the same supply is connected to both pins.

LorenzMeier, 2016-01-24:
No, there would be a massive problem. @nickarsov If you connect the 5V rail you can fry Pixracer if a servo is back-driven (which happens on planes and servo gimbals).

Building drones is an art. The seemingly simple is quite complex to get right.

LorenzMeier, 2016-01-24:
@pkocmoud FYI: You need to ensure your S.PORT cables do not connect the 5V line.

LorenzMeier, 2016-01-24:
We have seen quite a few people connecting servos to the receiver even if connected to the autopilot.

kd0aij, 2016-01-24:
@LorenzMeier Thanks, I see now they are 2 different supplies, and that this would tie the VDD_5V_RECEIVER and VDD_5V_PERIPHERAL supplies together, disabling the back-emf protection provided only for the RX supply.

nickarsov, 2016-01-24:
Don't forget TVS103 !

pkocmoud, 2016-01-25:
The manual is updated.