pixhawk/Hardware #57: Pixracer OEM

Naish88, 2016-01-27:
Hi, 
I will like to suggest a new derivative of the Pixracer.
Many people and companies now are building they own mini drones, and for tight integration they want to use the AP as a daughterboard soldered on top of main distribution board. This reduces the size/weight and increases reliability, plus is easy to integrate into products for commercial companies. In concept is like the Pixhawk 2 cube or the VRbrain core. 
The main difference, will be the removal of all the connectors, and the use of one side only of the PCB. SD and USB socket can be optional soldered on the underside, or just installed outside the AP module. 

Pixhawk 2 won't be available for sale, and VRbrain is closed sourceHW and works on a separate branch of ardupilot/px4, so not ideal. 
Anyone else interested in the Idea? I'm happy to work on it.

pkocmoud, 2016-01-27:
Naish88,

Please feel free to start something. I don't think this is the proper place to propose this as it is not an issue. While Nick and I are too busy at the moment, we could contribute in the future.

Naish88, 2016-01-28:
Philip, 
Sorry If I posted into the issue section. I was thinking that's the best place where Leonard Nick and you could all read this. Feel free to close this issue. 
In regards of the Pixracer OEM, I'm happy to take on the work. I just wanted to verify that everyone is happy and no-one feels like I'm coping someone's design. I will be really glad if any of you will like to help  with suggestions or anything else, also will be really great if Nick will be able to manufacture and sell the boards, as he has already the Pixracer in the pipe. 
Also there is any chance I can get the Altium files? Will save me heaps of time, instead of redrawing schematics and verifying footprints. a Github section for the Pixracer Hw will be great.

proficnc, 2016-01-28:
Pixhawk 2 WILL be available for sale!  

I am working to secure supplies.

Contact me if you have any questions

closed 2016-11-07 by pkocmoud