pixhawk/Hardware #58: No GPS -  Pixhawk defective ?

Tesa99, 2016-01-29:
Hello,
suddenly the pixhawk stop working with GPS from 3DR . In mission planner I can see NO GPS, and it do not change by the time.

I have tried another GPS Ublox M8N and I have the same problem. 

Is there a procedure I can find out if Pixhawk or the gps modules are defective ?
Compas is working on both.
Thanks,
Radek

subzero10, 2016-07-02:
Hi, did you solve your problem? I have the same issue. I have an APM 2.6 and a 3DR Pixhawk. The GPS works on the APM but not on the Pixhawk. I want to be sure that everything else is the way it should be (cable, configuration), before attempting to replace my Pixhawk.

pkocmoud, 2016-11-07:
This seems like an isolated local hardware failure.

closed 2016-11-07 by pkocmoud