pixhawk/Hardware #60: TXS0108QFN20 bi directional level converter problem with emax escs

dhirajdhule, 2016-02-05:
The old problem with emax and some other escs ([check this thread](http://diydrones.com/forum/topics/getting-pixhawk-to-work-with-emax?id=705844%3ATopic%3A1912186&page=2#comments)) with pixhawk. 
These esc's are loading the level converter and pwm output signal level decreases from 3.3V to ~2.2 V. 
The  TXS0108QFN20 seems to have little high output impedance >4K which is creating problems specifically with EMAX escs. Even though Emax guys have admitted that their escs draw extra current, we can look for some other options to use as buffer on pwm output channels. The problem persists even if the escs are not powered which also is obvious.
I tested the controller output before the TXS0108QFN20  and its normal while the buffers output is loaded. Anyway a good signal buffer for output needs very low output impedance.

proficnc, 2016-02-06:
If you look at the latest Pixhawk 2 drawings, you will see this issue has been rectified

closed 2016-02-06 by proficnc

Mr-Anderson, 2016-08-16:
I looked at the drawing and could not find a change that would rectify this issue. Could you please point out the change? EMAX is only using a 10K pull down on their signal line which does not seem unreasonable. The currents drawn by the ESC at 3.3V are only 330uA. These ESC's work with almost every other receiver and flight controller on the market. It seems like the PixHawk drive strength is awfully low for an external signal.

proficnc, 2016-08-16:
The TXS has been replaced.  And the resistor values dropped to 120 ohm.

However, I have never seen an installation where this occurs if the ESC is correctly wired.

It is critical that both the ground and signal pin are connected.

MuneebaAsif004, 2022-03-25:
Does anyone know what has TXS been replaced with?

davids5, 2022-03-25:
From FMU V4 and above the TXS0108QFN20 were not used, just 220R series
resistors and TVS. See the last page of
https://github.com/pixhawk/Pixhawk-Standards/blob/master/DS-010%20Pixhawk%20Autopilot%20Bus%20Standard.pdf

The pattern is

Inboard -----220R---+---------------- connector

                                TVS

                                  +

                                  GND

*From:* MuneebaAsif004 ***@***.***>
*Sent:* Thursday, March 24, 2022 5:24 PM
*To:* pixhawk/Hardware ***@***.***>
*Cc:* Subscribed ***@***.***>
*Subject:* Re: [pixhawk/Hardware] TXS0108QFN20 bi directional level
converter problem with emax escs (#60)

Does anyone know what has TXS been replaced with?

—
<https://github.com/pixhawk/Hardware/issues/60#issuecomment-1078527717>, or
unsubscribe