pixhawk/Hardware #65: in PX4FMUv2.4.5.pdf  PA10 should be labelled USART1_RX not USART1_TX

kwikius, 2016-05-13:
In sheet labelled shet 1 of 12, top left I think PA10 can only be USART1_TX, but is labelled USART1_RX

https://github.com/PX4/Hardware/blob/master/FMUv2/PX4FMUv2.4.5.pdf

davids5, 2016-05-13:
Not sure what you are looking at....

proficnc, 2016-05-13:
Page 1 is the FMU... It is the RX for the IO, and that is exactly how it is labeled

closed 2016-05-13 by proficnc

kwikius, 2016-05-13:
Sorry .. Clearly either I am an idiot or the labelling could be clearer :)

davids5, 2016-05-13:
No worries...It has been an issue with RS-232 (EIA) since day 1. If I was "EIA" I would have called it TX_TO_ME and RX_FROM_ME :)

proficnc, 2016-05-13:
That's why I love the SPI method... Master in slave out.. MISO etc...

Every board I lay out, I always panic on this stuff...

That bit could do with an additional net Kable to clear it up.