pixhawk/Hardware #67: ? pixhawk hardware issue or improvement ?

drones-squad, 2016-08-01:
Hy all,
I have recently noticed something strange about the pixhawk 2.4.5 hardware.
One of the sensor on the pixhawk is called a LSM303D ( a 3D magnetometer and accelerometer)
I was looking at the pixhawk schematics that are to be found on github HERE and saw that the sensor surrounding parts are different then what it's manufacturer suggest. ST shows a completely different arrangement for this. I was just wondering why did the people who build pixhawk reduce this surrounding parts or replaced them with other values. I wanted to hear the explanation of this and how it affects or improve the flight controller.
[image]

Please check the file in attachment. 

Greets

apache405, 2016-08-02:
There is a thread for this on drones-discuss as well.

drones-squad, 2016-08-02:
Hello apache405,

yes but nobody seem to answer ... I contacted David Sidrane and asked me about this issue and suggested to post this here on github. He also has not received any information for now as I am aware 

greets

LorenzMeier, 2016-08-10:
The values were chosen to match the requirements of the 2011 version of the data sheet and generalised to reduce the number of values per board. You might want to update the schematic to match the last manufacturer info.

Most importantly you will want to upgrade the sensors, as sensors available in 2011 are now quite outdated.

udawat, 2016-08-10:
A good combination of sensors from ST include the following:
LSM6DS33 (3D Accel + 3D Gyro) as an update to the L3GD20(H) Gyro.
LSM303D (3D Mag + 3D Accel) - Better than standalone magnetometers.
Both support SPI at 10MHz.

Honeywell's HMC5983 is now End of Life part. For a 3D Magnetometer only option, LIS3MDL is there.

LSM9DS1 should not be considered as it is just two sensor dies in a single package (similar to MPU9250 and 9150).

For Pressure Sensor, Bosch BMP280 can be an option. It supports SPI  at 10 MHz. But, its resolution is 20 bits (not sure) as compared to the 24-bit ADC of MS5611.

drones-squad, 2016-08-10:
Hello Lorenz, 

You say that the values were chosen to match the 2011 requirements, can you confirm that this issue was also to be found on the more recent boards like 2.4.6 ? If one will change values then I suppose it will have an impact on the sensor / firmware as well ... ?

And can you please perhaps let me know what kind of sensors you are talking about ? I have searched for example for a good mpu6000 alternative but none found. The newer 9060 or 9050 or whatever are all based on the mpu6000 but with far more less capabilities and functions as they are also cheap. The REAL mpu6000 for example is a rather reliable but expensive part. The lsm303d does all not too bad and the L3GD20H is newer then the L3GD20. But then again Please advice if I missed something, there are a dozen of manufacturers out there with thousands of parts.

greets

igalloway, 2016-08-10:
I haven't done a full competitive analysis yet, and there certtainly are
differences (I2C vs SPI) but NXP mix would be:

·        FXOS8700 Mag + Accel with up to 400Hz sampling rate in hybrid mode
(I2C based)

·        FXAS21002 Gyroscope with up to 800Hz sampling rate

·        MPL3115A2 Pressure Sensor/Altimeter with typical resolution of
0.3m at 1Hz

·        Free/Open Source 9-axis sensor fusion for computing orientation