pixhawk/Hardware #74: Altium project for PIXRACER R12 references wrong source files

ajawamnet, 2017-01-28:
It seems to reference -R09 files

ajawamnet, 2017-01-28:
[image]

pkocmoud, 2017-01-30:
@ajawamnet 

Sorry about that, I originally renamed the files and that was not the proper procedure. I have properly fixed the file names and paths in the project file.
 
https://github.com/AUAV-OpenSource/FMUv4-PixRacer/

closed 2017-01-30 by pkocmoud

ajawamnet, 2017-01-30:
If you ever need any help with the Altium silly, let me know.
The Project Packager is what I typically use for releasing. They now 
have the vault stuff and all, but it kinda sucks.

Even for sex toy (Google my name - there's a Mother Jones article).

Sanjay and I (see my website) have been using (kinda more like battling) 
those MEMS things since they came out. He did some stuff for ESA back 
when the MEMS stuff was first introduced. That and RTK with dual GPS.

  I recall my brother telling me about his work on the guidance ring for 
the Saturn V Block II booster and dealing with the ST-124
https://en.wikipedia.org/wiki/ST-124-M3_inertial_platform

There's one at Udvar Hazy/Smithsonian - the curator was jokingly asking 
me if he still had an LVDC 4 bit lying around...
https://en.wikipedia.org/wiki/Saturn_V_Instrument_Unit
https://en.wikipedia.org/wiki/Saturn_Launch_Vehicle_Digital_Computer

-- 
WAM
http://www.ajawamnet.com
http://www.ajawamnet.com/ajawamnet/marketturd.htm
------------

More information on my lab and shop:
http://www.ajawamnet.com
http://www.ajawamnet.com/ajawam3/swarf/swarf.htm
------------------------
"Success is going from failure to failure without a loss of enthusiasm."
~~ Sir Winston Leonard Spencer-Churchill (1874-1965)

"I've failed at so many things so many times I make Abraham Lincoln look like Bill Gates"
~~ Wayne Mitzen (1959-  )

"What makes an expert isn't so much what they know,
It's that they've done similar things so many times wrong
They know what not to do"
~~ Wayne Mitzen (1959-  )

"You just keep pushing. You just keep pushing. I made every mistake that could be made. But I just kept pushing."
~~ Rene Descartes 1596-1650

"I hate the way people just go from ignorance to despair so quickly; it's the lazy man's option"
~~Professor Tim Flannery (1956-  )

"When you do things right, people won't be sure you've done anything at all."
~~ God, in Futurama episode "Godfellas"

"Two things are infinite: the universe and human stupidity;
and I'm not even sure about the universe."
~~ Albert Einstein (1879-1955)

"Some scientists claim that hydrogen, because it is so plentiful, is the basic building block of the universe. I dispute that. I say there is more stupidity than hydrogen, and that is the basic building block of the universe."
~~ Frank Zappa (1940-1993)

On 1/30/2017 8:05 AM, Phillip Kocmoud wrote:
> @ajawamnet
>
> Sorry about that, I originally renamed the files and that was not the proper procedure. I have properly fixed the file names and paths in the project file.
>   
> https://github.com/AUAV-OpenSource/FMUv4-PixRacer/
>