pixhawk/Hardware #77: Zubax Sapog LED package reversed polarized.

taskx6004, 2017-06-08:
I have build a small batch Sapog ESC according to the eagle files, flashed, motor beeps, but the four color LED is not working. after checking polarity i found the original Sapog LED was reversed and hand solded. please check your sch or please update the design library.

or my fab did it wrong, i will double check.

It's a amazing project @pavel-kirienko. Thanks for you effort. I will not sell these boards for profit. I would like to donate samples to people who will investigate how to use SPI-CAN adaptor to get it work on linux, RPi especially.

pavel-kirienko, 2017-06-08:
@taskx6004 thank you for the kind feedback. Make sure to also check out the new reference manual we're creating: https://files.zubax.com/products/io.px4.sapog/Sapog_v2_Reference_Manual.pdf

Calling in @j3qq4hch. Alex, could you verify this please? The LED works fine on the production Zubax Orel 20, perhaps the open sourced design has a mistake in it, or the @taskx6004 has soldered it incorrectly indeed.

j3qq4hch, 2017-06-08:
I checked it again. Zubax Orel  and sapog reference have identical led connection. I guess, fab soldered it wrong. We have already seen this before

pavel-kirienko, 2017-06-08:
Thanks, Alex. Nothing unheard of indeed, PCB vendors have a thing with this type of LED.

closed 2017-06-08 by pavel-kirienko