pixhawk/Hardware #78: Pixhawk LED Module Chip?

tubeme, 2017-07-07:
Hello guys,

We are trying to design our own LED module for the Pixhawk controller. As far as we understand there is a chip on the board that recieves information from I2C bus then accordingly controls the led. Some sort of LED driver with I2C control.

I suppose there is firmware that is doing the math?

So we've had a LED module that we like but it uses a chip that is no longer available. It is out of production etc. It is Toshiba chip TSA62724.

So my question is there any other chip that can be used for the LED module to drive the leds from I2C?

Thank you in advance.

LorenzMeier, 2017-07-07:
Let me have a look. I would be happy to add support for a new model.

igalloway, 2017-08-05:
It's not clear what the overall goal is with this new "LED Module". I just thought i would mention a couple of parts, as a flexible options for driving both a few (RGB) and many LEDs:
Hopefully helpful?

[NExperia NPIC - Shift register based, cascade-able](https://www.nexperia.com/products/logic/family/NPIC/#/p=1,s=0,f=,c=,rpp=,fs=0,sc=,so=,es=)

[NXP Selector guide](http://www.nxp.com/products/interfaces/ic-bus-portfolio/ic-led-controllers:MC_48878#/&page=2)

[NXP I2C based - PCA9634](
http://www.nxp.com/products/interfaces/ic-bus-portfolio/ic-led-controllers/8-bit-fm-plus-i2c-bus-led-driver:PCA9634)

[NXP I2C based - PCA9551, Fast mode with Programmable blink rates](http://www.nxp.com/products/interfaces/ic-bus-portfolio/ic-led-controllers/8-bit-i2c-bus-led-driver-with-programmable-blink-rates:PCA9551)

tubeme, 2017-08-05:
Thanks a lot guys!! Appreciated @LorenzMeier @igalloway

tubeme, 2017-08-06:
@LorenzMeier @igalloway If we use one of these chips do we need a driver for it or it is in PX4 already?

igalloway, 2017-08-07:
Can you explain a bit more what you are trying to do?
I'm assuming this is either some kind of remote status indicator?

There are external SPI ports on most PX4 hardware.
There are software drivers for SPI communications
To actually control and illuminate the LED(s) in whichever way you want
would require some software written both to control the SPI device and then
control the LED(s) in the way you want.

tubeme, 2017-09-21:
@igalloway We are going to use it for a Status Leds for the PX4. So probably we have to implement it ourselves because there is no chip that could be used right out of the box for now. The one's used are discontinued.

a146860, 2018-08-14:
What replacement