pixhawk/Hardware #82: help ! px4io not found error

mehrdadshiri, 2017-09-12:
hi every body
i was flying my pixhawk on a quad-copter for the first time and suddenly an alarm came off and and it landed.
so I connected the usb right after and now it says px4io not found in mission planer and the on the Qground control it says "cannot communicate"  . i have read on the net that some people suggest to start with the safety switch pressed for firmware restart and it didn't help at all and the switch is not blinking anymore. 
has anyone had a same experience before or have any suggestions to fix this ?

hamididari, 2018-03-17:
hi
can you check the px4io micro is health or not (connect to px4io via st link and read memory)

pkocmoud, 2018-05-22:
@mehrdadshiri - Please seek assistance on this issue from your hardware supplier or the flightstack software's discussion board. http://discuss.px4.io/ for PX4.

closed 2018-05-22 by pkocmoud