pixhawk/Hardware #95: How to connect pixhawk 2.4.8 to Nvidia Jetson Nano?

atmadeep, 2019-08-20:
Hi, I'm working on making a semi autonomous drone using vision based methods. Has anyone tried connecting a pixhawk 2.4.8 with jetson Nano? Any help would be appreciated

dogmaphobic, 2019-08-20:
[image]

The UART pins are separate for easy reach.

atmadeep, 2019-08-20:
Have you tested this over ROS? Can you get the messages in mavros?

dogmaphobic, 2019-08-20:
Yes, there isn't really any difference than running mavros on an RPI or an Intel NUC (running a Debian Linux).

atmadeep, 2019-08-20:
Thank you very much.

atmadeep, 2019-12-02:
New problem here, Every time I connect jetson nano to pixhawk, it turns off and won't start again until disconnected. I'm not connecting the red/power cable. 
What can be the problem now?

pkocmoud, 2019-12-02:
@atmadeep This is really not a place for usage assistance. You would have better luck in one of the discuss forums.

Depending on the flight software you are using you might try: 

PX4: https://discuss.px4.io/
Ardupilot: https://discuss.ardupilot.org/

proficnc, 2019-12-02:
Make sure your voltage levels are correct on your io

busradurak, 2020-06-19:
> New problem here, Every time I connect jetson nano to pixhawk, it turns off and won't start again until disconnected. I'm not connecting the red/power cable.
> What can be the problem now?
Try to give power only the nano if you connect to power both of them, some problem might be occured.

oguzhanonder38, 2021-07-27:
> > New problem here, Every time I connect jetson nano to pixhawk, it turns off and won't start again until disconnected. I'm not connecting the red/power cable.
> > What can be the problem now?
> > Try to give power only the nano if you connect to power both of them, some problem might be occured.

Hello busra have you solved connection problem? If you solved the problem, please connect to me. Thx. : *

reinzler, 2023-06-05:
Hi, guys,  I usually have same problem, if Pixhawk had been connected to Jetson Nano before I powered on the Nano, Nano cant detect Pixhawk until I physically disconnect/ connect MicroUSB cable, Did someone find an answer?

hamishwillee, 2023-06-05:
Not the best place for this question - https://github.com/pixhawk/Hardware/issues/95#issuecomment-560539815

But if you find an answer cross post.

closed 2023-06-05 by hamishwillee