pixhawk/Hardware #96: Accelerometer calibration fail

RandomHacksYoutube, 2019-10-29:
Hi, I have an PX4FMUv2.4, when I try to calibrate the accelerometer in mission planner it asks me to level the pixhawk, position left, and right and then it fails, every single time, no matter how still I am during this proccess. I already replaced MPU6000, L3GD20 and LSM303D but it made no difference. I don't even know how to debug this issue, can somebody shed some light on this? Is there a way to know which of these 3 ICs is causing the problem or if it is something else?

pkocmoud, 2019-10-29:
Hello @RandomHacksYoutube , this GitHub project issue section is more suited to fundamental design flaws. You will want to head over to https://discuss.ardupilot.org/ for assistance with this sort of issue.

RandomHacksYoutube, 2019-10-29:
I believe this is a hardware issue so I thought this would be the place, but I will head over there then. Thanks.

pkocmoud, 2019-10-29:
I understand but the software project's support could help you understand which sensor is causing your trouble by reviewing your logs. The Pixhawk design is fundamentally sound and it seems most likely the issue you are experiencing is specific to your board.