pixhawk/Hardware #111: FMU 5x/6x Baseboard

seankovacs, 2022-06-26:
Hello,
I'm attempting to create a custom carrier/base board for the FMUv5x/6x module. Reviewing the Autopilot Standard which HolyBro seems to be the only one support (!?), I initially was under the impression that the base/carrier board was simply a companion that hosted external ports that feed the main module. After further review, it seems the baseboard hosts a IO MCU and EEPROM (among other simple circuits). 
Few questions if you don't mind:

- How come there isn't any documentation on the baseboard IO MCU pinouts? How do the pins map to the Main MCU?
- Since there are existing controllers that don't have an IO MCU, I'm assuming you can create a baseboard without it?
- I see that there is a EEPROM on the baseboard? What/who programs it?

davids5, 2022-07-06:
> Hello, I'm attempting to create a custom carrier/base board for the FMUv5x/6x module. Reviewing the Autopilot Standard which HolyBro seems to be the only one support (!?), I initially was under the [...]

The IO is optional. But there has not been a board done for the VxX that did not use it. 

This is a serial connection. Referring to  https://github.com/pixhawk/Hardware/blob/master/FMUv2/PX4FMUv2.4.5.pdf Sheet 8
will get you the wiring. @mrpollo is there more docs you can share with  @seankovacs? 

> * Since there are existing controllers that don't have an IO MCU, I'm assuming you can create a baseboard without it?

Yes.

> * I see that there is a EEPROM on the baseboard? What/who programs it?

Ethernet configuration and device manifest 

See https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v5x/src/mtd.cpp#L41 and 
https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v5x/src/mtd.cpp#L69-L82

The application `netman` manages the Ethernet configuration.

Device manifest will be used in the future for a "device tree" like description of the devices connected. It is a versioned data struct and currently in used to expand the HW VER ID when the R1/R2 is set to code point 7. It is programed by the MFG when used.

seankovacs, 2022-07-06:
Thanks for your response @davids5 . So it sounds like the EEPROM is optional as well? Especially if the need for ethernet is not needed?

davids5, 2022-07-06:
If you are going to have a custom board and build of the firmware you can do what ever you like. But it will not be compatible with the exiting PAB V5X and V6X  FMUM's built by the current manufactures and will not be PX4 VxX compatible.

alekh1stzoom, 2023-01-27:
Does anyone you have the schematic for the FMU v6x?