pixhawk/Hardware #112: Several question about schematic of fmu 2 and 3

Trinity030, 2022-07-28:
Why Fmu mcu communicate with Io mcu series 220 ohm ?
for impedance matching ??

davids5, 2022-07-28:
In the old days, the IO could land a FW (not really feasible with MC) with Fail Safe manual RC control if the FMU died. 

So the 22R was for Isolation to limit the current incase of a catastrophic failure on the FMU SoC. 

In a recent PR the IO was lobotomized so the value add of the isolation is almost nothing.

davids5, 2022-07-29:
Firmware is here https://github.com/PX4/PX4-Autopilot or builds are under actions i.e.  https://github.com/PX4/PX4-Autopilot/actions/runs/2753822626

Schematics are here https://github.com/pixhawk/Hardware

proficnc, 2022-07-30:
This is still critical, in a PX4 FMzU power failure, the IO may possibly still be powered 

Also, though some forget... Pixhawk "Hardware" was an ARDUPILOT AND PX4 effort, and Ardupilot still has RC pass through which is a critical safety feature used by many many.