pixhawk/Hardware #114: Lost power on current main power brick causing the avionic reboot

hitagisol, 2022-11-02:
I notice that in the case when multiple power bricks present, if the current main power source failed the avionic will reboot.
Here is the test I've made on Pixhawk 5X:
 - Hardware setup: Connect Power 1, Power 2, USB as power source, Power 1 and Power 2 come from the Holybro PM03D module and the USB comes from my laptop.
 - Test procedure: 
    1. Remove the connector from Power 1, while the Power 2 and USB remains connected, the avionic goes into the reboot routine.
    2. Remove the connector from Power 2 while the Power 1 and USB remains connected, the avionic keeps working without reboot.
    3. Remove the connector from Power 1, trigger the avionic goes into reboot. After the reboot, remove the connector from Power 2, the avionic reboots again.
Right now I'm assuming this is caused by the path switch handled by the LTC4417 controller, but I have no idea what test I can perform to prove that.

hitagisol, 2022-11-04:
@LorenzMeier I'm sorry for the manually ping but since this repo is not so popular I don't know if there is someone taking issues here periodically.

davids5, 2022-11-04:
@hitagisol Have you verified this on more than one FMU?

hitagisol, 2022-11-05:
> 

@davids5 I have the same results on two different Pixhawk 5x. On the Pixhawk 4 board I do not have this issue. Currently I only have these three board in my hand.

vincentpoont2, 2022-11-07:
@hitagisol May I ask where you purchase this unit from? I have also sent you an email. Please check.